glenn20/micropython #1: Memory issue with AIOESPNow

luiz-brandao, 2022-04-19:
I started having memory allocation errors once I tried porting my code to use the new uasyncio module AIOESPNow. 
`MemoryError: memory allocation failed, allocating 4224 bytes`

This happens even if I have no espnow code and I just import the module in the context of my application like so: 
`import aioespnow as espnow`

If I comment the import line and run `micropython.mem_info()` I get this: 
`GC: total: 64000, used: 42480, free: 21520`, 

21520 bytes should be plenty no?

Without any code loaded, the REPL gives me this when running micropython.mem_info():
`GC: total: 64000, used: 6256, free: 57744`

If I then import aioespnow I get:
`GC: total: 64000, used: 12016, free: 51984`

Roughly 6kb doesn't seem like a lot, but if the import happens in the context of the application, it crashes and I have no idea why yet.

Any clues on what might be happening? I know that it's not much to start with but I appreciate any insights.

Thank you.

glenn20, 2022-04-20:
Thanks for the report. I would expect the [aioespnow.py](https://github.com/glenn20/micropython/blob/espnow-g20/ports/esp32/modules/aioespnow.py) module to have a pretty small memory footprint (but I might be wrong). It may well be that it is just enough to tip your application over the edge.

What is the memory consumption if you replace `import aioespnow as espnow` with:
```
from esp import espnow
import uasyncio
``` 
(these are imported by aioespnow.py and I imagine they would use substantially more memory that aioespnow.py)

Which device are you using? And which image? Your total GC RAM is signficantly smaller than for esp32 or esp32-s2.

You could try putting `gc.collect()` at various points after loading modules (esp. .py modules that have not been precompiled to .mpy) to be more pre-emptive with memory.

glenn20, 2022-04-20:
I tested memory usage on my (freshly rebooted) ESP32 device:
```
import aioespnow as espnow
```
consumes 3kb.

And executing the following code consumes a total 4kb (including the 3kb):
```
from esp import espnow
import uasyncio as asyncio
import aioespnow as espnow
e = espnow.ESPNow()
e.init()
```

In fact, loading espnow consumes 144 bytes of heap, uasyncio consumes 2496 bytes and aioespnow consumes 640 bytes.

It appears something in your app code tries to allocate 4224 bytes of contiguous storage and fails. As your memory becomes fragmented after loading your app it may be unable to allocate that buffer (and the little bit of extra RAM from the aioespnow load (640 bytes) may just tip it over the edge).

Some tips which might help (and you may already be aware of these ;-):

- Allocate larger memory buffers as early after booting as possible (before memory becomes fragmented), ie. reorg your code so that memory blocks are allocated immediately as soon as your app starts up.
- It can also help to insert some gc.alloc() calls at appropriate places in your app code.
- Pre-compiling your app code to .mpy files using mpy-cross can save memory fragmentation for the compilation phase when the .py modules are loaded.
- If you have a large app code you can freeze that into the micropython image to save RAM and reduce fragmentation (see https://docs.micropython.org/en/latest/develop/optimizations.html#frozen-bytecode).

Here is the memory trace of loading and init-ing aioespnow on my freshly rebooted ESP32 device (no apps loaded):
```
>>> import micropython
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 1792, free: 109376
 No. of 1-blocks: 34, 2-blocks: 9, max blk sz: 18, max free sz: 6825
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 1248, free: 109920
 No. of 1-blocks: 22, 2-blocks: 8, max blk sz: 18, max free sz: 6819
>>> import aioespnow as espnow
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 4848, free: 106320
 No. of 1-blocks: 114, 2-blocks: 22, max blk sz: 18, max free sz: 6633
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 4304, free: 106864
 No. of 1-blocks: 85, 2-blocks: 21, max blk sz: 18, max free sz: 6636
>>> e = espnow.ESPNow()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 5328, free: 105840
 No. of 1-blocks: 100, 2-blocks: 24, max blk sz: 18, max free sz: 6589
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 4752, free: 106416
 No. of 1-blocks: 88, 2-blocks: 23, max blk sz: 18, max free sz: 6600
>>> e.init()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 5584, free: 105584
 No. of 1-blocks: 99, 2-blocks: 24, max blk sz: 33, max free sz: 6567
>>> gc.collect()
>>> micropython.mem_info()
stack: 704 out of 15360
GC: total: 111168, used: 5264, free: 105904
 No. of 1-blocks: 89, 2-blocks: 22, max blk sz: 33, max free sz: 6567
>>> 
```

luiz-brandao, 2022-04-20:
> Which device are you using? And which image? Your total GC RAM is significantly smaller than for esp32 or esp32-s2

I'm using ESP32-WROOM-32E, when I run `gc.mem_free()` on a freshly rebooted device I get only `57920` and `micropython.mem_info()` gives me the following:

```
>>> gc.mem_free()
57920

>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 64000, used: 10144, free: 53856
 No. of 1-blocks: 238, 2-blocks: 35, max blk sz: 18, max free sz: 3356
```

What happened to the rest?

glenn20, 2022-04-20:
Odd. And what does mem_info() report if you use a vanilla v1.18 image?

luiz-brandao, 2022-04-20:
Indeed, after trying a vanilla v.1.18 image I got more free memory.

```
>>> import gc
>>> gc.mem_free()
95648
>>> import micropython
>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 111168, used: 21808, free: 89360
 No. of 1-blocks: 476, 2-blocks: 72, max blk sz: 101, max free sz: 5572
>>> 
```

With a the image firmware-esp32-GENERIC-idf44-v1.18 I got less memory
```
>>> import gc
>>> gc.mem_free()
50112
>>> import micropython
>>> gc.mem_free()
44992
>>> 
```

I also tried with firmware-esp32-GENERIC-idf402-v1.18, and with that one I got more memory

```
>>> import gc
>>> gc.mem_free()
97280
>>> import micropython
>>> micropython.mem_info()
stack: 736 out of 15360
GC: total: 111168, used: 19008, free: 92160
 No. of 1-blocks: 433, 2-blocks: 61, max blk sz: 101, max free sz: 5747
>>> 
```

glenn20, 2022-04-22:
Hmm - it seems that micropython releases use v4.2 IDF for esp32 and v4.4 for esp32s2/s3/c3 images (I think). I'll adopt that same approach for my future builds. I am **guessing** that the memory issue is triggered by some update to the v4.4 IDF (I updated recently for that firmware-esp32-GENERIC-idf44-v1.18.bin).

closed 2022-04-22 by glenn20

glenn20, 2022-04-23:
@luiz-brandao : Hopefully the [new espnow-g20-v1.18 images](https://github.com/glenn20/micropython-espnow-images/tree/main/20220423_espnow-g20-v1.18-14-g78cdcdfdc) I have uploaded include the i2c fix and the memory fix too. Good luck. Ive tested the memory fix on my devices and all seems OK.