glenn20/micropython #2: Can't import name TouchPad on Lolin S2 image

jarodriguez-itsoft, 2022-04-21:
Hi.
I'm using latest image to test the ESP Now functionality (great work btw!), but I found I get an error when trying to use the TouchPad functionality.
I get "ImportError: Can't import name TouchPad" when using
`from machine import TouchPad`

Image is LOLIN_S2_MINI-20220117-v1.18.bin

Thanks

glenn20, 2022-04-22:
Hmmm. The espnow PR makes no changes to the config options that would affect the TouchPad.

I just checked [machine_touchpad.c](https://github.com/micropython/micropython/blob/master/ports/esp32/machine_touchpad.c) in the micropython master branch and the entire code is wrapped in "#if CONFIG_IDF_TARGET_ESP32" which means the touchpad is NOT compiled in on the S2 or S3 by default. I can see that there is code to suport the S2 and S3 in the machine_touchpad.c, but it is not enabled by default.

Can you verify if the TouchPad is enabled on vanilla micropython v1.18 images? It seems to have been deliberately disabled - perhaps because the S2 and S3 support is not yet mature.

glenn20, 2022-04-22:
Actually, I just checked with a v1.18 micropython image on my Feather S2. The TouchPad is also not available on the v1.18 micropython releases.

I will close this issue for now. You may want to post an issue on upstream micropython (if there is not one already) - or inquire about the status of the S2 TouchPad support on the forum. Good luck.

closed 2022-04-22 by glenn20