glenn20/micropython #4: Developing ESP-NOW for circuitpython

americoperez49, 2022-09-20:
Hi Glenn, I wasn't sure how else to get a hold of you so sorry for the "issue".

I am interested in integrating ESP-NOW with CircuitPython but I have no idea where to start. 

Would you be interested in helping me get started/maybe coach me a little so I can attempt this?

glenn20, 2022-09-27:
@americoperez49  Sorry for the slow response - I've been on holiday for the last week. I know little about circuitpython - especially in the C layer, but I'm happy to help out with this. I _imagine_ it would not be too painful ;).

Do you have a good understanding of the circuitpython internals?

Based on the little that I do know, I believe there is strong compatibility at the C layer (types, etc.) but differences in driver implementations. The esp-now module does lean on the WLAN class implementation and may be sensitive to differences or divergences there.

americoperez49, 2022-09-27:
Hi Glen,

Not a problem. Hopefully your holidays are going well ☺️.

Honestly, my knowledge of the lower level implemention of circuitpython is
next to none. With that said, I've been programming for a long time now and
I'm sure that whatever I find I'll be able to figure out either on my own
or by asking the circuitpython devs directly.

closed 2023-04-01 by glenn20