glenn20/micropython #5: docs:

krishnak, 2022-09-27:
You have mentioned about broadcast messages between ESP8266 and ESP32 not appearing unless they are on AP_IF.

I have noticed this today
Sender ESP32 broadcast address 
broadcastpeer = b'\xff' * 6
e0.add_peer(broadcastpeer,ifidx=network.AP_IF)

Any broadcast sent from ESP32 on both the interfaces are not received by ESP8266, please confirm whether you are able to reproduce this and update the document accordingly.

krishnak, 2022-09-27:
Apparently this is a BUG in the underlying SDK and it appears to have been fixed in Arduino implementation, you may need to check with micropython team
https://github.com/esp8266/Arduino/issues/6174

krishnak, 2022-09-28:
Just an update I have download the ESP-RTOS-SDK for esp8266  - an app built using that in C, can receive and send broadcast messages from ESP8266 it appears that the micropython team are using an older probably a non OS SDK which has bugs (including the DDoS loop hole) which is crashing my ESP8266 WiFi after 30 seconds of connecting (only to my router) a separate thread is going on regarding this on micropython discussion board .

I think they need to upgrade the ESP SDK on micropython

glenn20, 2022-10-02:
Thanks for following this up. This issue is noted in the [online docs](https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html#broadcast-and-multicast).

Yes, micropython uses the older non-os sdk for esp8266 and while that is the case there is not much I can do to resolve this issue. I'm not aware of any attempts underway tp adopt an updated SDK for esp8266 for micropython.

closed 2022-10-02 by glenn20