glenn20/micropython #6: ESP Now LR mode

krishnak, 2022-10-03:
When you enable Long range mode w0.config(protocol=network.MODE_LR) - I am not longer able to see the WiFi beacon of AP from my computer's network scan. Can you please confirm whether you can connect to AP with LR mode enabled?

krishnak, 2022-10-03:
or is there an option to have LR and WiFi enabled simultaneously?

glenn20, 2022-10-08:
You **should** be able to do:
```python
w0.config(protocol=network.MODE_11B|network.MODE_11G|network.MODE_11N|network.MODE_11N)
```
and wifi will work as well as LR mode. From my understanding LR mode will be used if the other device is also capable of LR mode - but I have not verified that myself. (see [esp_wifi_set_protocol()](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/network/esp_wifi.html#_CPPv421esp_wifi_set_protocol16wifi_interface_t7uint8_t)).

The protocol is a bitmask you can use to select which modes you want enabled.

I have not personally worked with LR mode much at all - other than when prompted to check a few things by users :-).

krishnak, 2022-10-08:
No Luck, the multiple modes when include LR - the beacon disappears.When  LR mode is in the list only ESPNow works

krishnak, 2022-10-08:
I will revisit this LR mode after finishing my target application. What I have observed practically so far is, with two devices in LR mode we don't gain anything extra. This is wit ESP-IDF SDK itself

I get reception on the receiver upto -98dB RSSI without LR mode itself, which is good enough for time being.

mcccol, 2022-10-11:
LR is a different physical layer encoding to normal WIFI. Your best bet
would be to have another esp32 as an AP, and somehow connect it up to your
WIFI router. Ethernet, perhaps? That's an exercise for the reader

tyeth, 2024-02-11:
I'm very interested in this for CircuitPython. From my understanding LR mode can be enabled in addition to the b/g/n modes (presumably in some time-slicing sense), but I've seen problems if the wifi + esp-now channels vary, and another problem means esp-now is channel 1 on circuitpython for now.

 The other issue is it's probably unrealistic if you expect / activate the [true long range functionality (LowRate mode)](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/network/esp_wifi.html#_CPPv4N15wifi_phy_mode_t16WIFI_PHY_MODE_LRE), as I was under the impression that longer range [reduces the bandwidth](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/network/esp_wifi.html#_CPPv415wifi_phy_rate_t) to 500kbps/250kbps [presumably the LORA low-rates?] rather than the usual Mbps from >=WIFI_11B, so you'll want all the air-time available for low-bandwidth long-range comms instead. 

Maybe as suggested having a bridge device, or a co-processor like the matter devices do, would be best option. Personally I'm going for time slicing, manually if necessary (the nodes can repeat for a minute or so while the "hub" fires data over 802.11g for a few seconds).

This comment is a third "my 2cents", a third just a subscription to the topic, and a third a request for advice or question to see if you lot in micropython land have the same problems/restrictions.

glenn20, 2024-02-12:
> I'm very interested in this for CircuitPython. From my understanding LR mode can be enabled in addition to the b/g/n modes (presumably in some time-slicing sense), but I've seen problems if the wifi + esp-now channels vary, and another problem means esp-now is channel 1 on circuitpython for now.

Although the docs from espressif suggest you can mix the modes, I havent seen any reports that this works effectively with espnow and it is better to just select the LR mode as the only option. I suggest then trying the mixed modes.

Im not familiar with the circuitpython interface to espnow, but espnow traffic just runs over whatever channel the radio is currently set to. Eg. if you connect to an AP on STA_IF, the radio switches to the channel of the AP, and espnow traffic will be sent and received on that channel. If you start an AP on your device on channel X, then espnow traffic will be sent and received on channel X. (Caveat: espressif may have changed some of this behviour in the IDF, but my experience has been that there is no automatic swiching of channels between STA_IF, AP_IF and espnow operations.

So, if you start espnow, you can change the channel of operations by changing the channel on the STA_IF or AP_IF. Micropython allows manually setting the channel number of the STA_IF or the AP_IF, however it is possible that circuitpython only provides access to setting the AP_IF channel. (Earlier versions of the IDF did not support setting the channel on the STA_IF). Eg. turn on STA_IF, turn on AP_IF, set the channel of the AP and then turn off AP_IF. The radio will continue to operate on the channel you set for the AP_IF. (See the [micropython espnow docs](https://docs.micropython.org/en/latest/library/espnow.html#espnow-and-wifi-operation).

> The other issue is it's probably unrealistic if you expect / activate the [true long range functionality (LowRate mode)](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/network/esp_wifi.html#_CPPv4N15wifi_phy_mode_t16WIFI_PHY_MODE_LRE), as I was [under the impression](https://github.com/espressif/esp-idf/issues/12221#issuecomment-1765574374) that longer range [reduces the bandwidth](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/api-reference/network/esp_wifi.html#_CPPv415wifi_phy_rate_t) to 1Mbps/500kbps/250kbps [presumably the LORA low-rates?] rather than the usual Mbps from >=WIFI_11B, so you'll want all the air-time available for low-bandwidth long-range comms instead.
> 
> Maybe as suggested having a bridge device, or a co-processor like the matter devices do, would be best option. Personally I'm going for time slicing, manually if necessary (the nodes can repeat for a minute or so while the "hub" fires data over 802.11g for a few seconds).

This looks like a good starting point and then progress from there. One comment is that I have seen  that users often expect to see an improved RSSI reading in LR mode, but that is not how I believe LR mode operates. Instead it should result in a lower packet loss rate compared to wifi for the same (low) RSSI. (eg, see https://github.com/glenn20/micropython-espnow-images/issues/20#issuecomment-1207315635).

> 
> This comment is a third "my 2cents", a third just a subscription to the topic, and a third a request for advice or question to see if you lot in micropython land have the same problems/restrictions.

I am sure that we have some of the same restrictions, but over the last few years a few PRs have been merged into micropython (mostly in `network.WLAN.config()`) that use new features in the IDF that espressif has provided to better support espnow. Im not sure if those are available in circuitpython.