glenn20/micropython #8: No module names `_espnow`

AmirHmZz, 2022-11-29:
I've opened a PR from this branch to master branch of my own [fork of micropython](https://github.com/agroture/micropython) and then I've merged it. I've built it esp8266 successfully but I cannot import espnow because of this error:

```Python
Traceback (most recent call last):
    File "<stdin>", line 1, in <module>
    File "espnow.py", line 4, in <module>
ImportError: no module named '_espnow'
````

AmirHmZz, 2022-11-29:
What am I doing wrong? Is there any special step in building Micropython with your PR being merged?

AmirHmZz, 2022-12-02:
@glenn20 The same thing happens when I build [espnow-g20](https://github.com/glenn20/micropython/tree/espnow-g20) branch of your repo.

AmirHmZz, 2022-12-02:
Problem solved by adding this to `mpconfigboard.h`.
```
#define MICROPY_ESPNOW           	  (1)
```

I think it should be better to notice that espnow is not enabled by default in esp8266 builds.

closed 2022-12-02 by AmirHmZz

glenn20, 2022-12-03:
Sorry for the slow response - busy time. On esp8266, the following lines in `ports/esp8266/Makefile:45` *should* set MICROPY_ESPNOW to be enabled by default:
```bash
# Support ESPNOW by default on the ESP8266
MICROPY_ESPNOW ?= 1
ifeq ($(MICROPY_ESPNOW),1)
CFLAGS_MOD += -DMICROPY_ESPNOW=1
endif
```
It's possible some conflict with your merges or a recent update to micropython:master has caused this to stop working.

glenn20, 2022-12-03:
Actually, a change was committed to micropython master on October 11 that makes CFLAGS_MOD redundant. If you are building with a branch built against master since then, that would be the source of the problem. I will update my branch accordingly.

AmirHmZz, 2022-12-04:
And thank you for this nice work.