glenn20/micropython #9: Receiving messages from esp8266 with ESP32 connected to STA

AmirHmZz, 2022-11-29:
According to your documentation :

> Receiving messages from an ESP8266 device: Strangely, an ESP32 device connected to a wifi network using method 1 or 2 above, will receive ESP-Now messages sent to the STA_IF MAC address from another ESP32 device, but will reject messages from an ESP8266 device!!!. To receive messages from an ESP8266 device, the AP_IF interface must be set to active(True) and messages must be sent to the AP_IF MAC address.

I've tested it with an **esp01** device (which integrates **ESP8266EX** SoC) as sender and ESP32 as receiver and successfully received messages which are sent from esp01 to esp32 device. I should also mention that I've sent them to broadcast mac address.

glenn20, 2022-12-03:
Thanks for the report. That is an interesting observation that broadcast messages are received. I'll do a little more investigation and update the docs to clarify.

ramiws, 2023-03-26:
Instead of writing strangely why not report the bug to espressif? 
I can't believe the amount of people seeing the bug and not reporting it.
I report it espressif admitted the bug and provided me with hotfix blob of the espnow library.

glenn20, 2023-03-27:
> Instead of writing strangely why not report the bug to espressif? I can't believe the amount of people seeing the bug and not reporting it. I report it to espressif myself last week. They admitted the bug and provided me with hotfix blob of the espnow library.

That's interesting. I have reported a few issues to espressif over time, but I confess it seems I had incorrectly assumed this was an issue on the ESP8266 sdk - which espressif had previously told me doesn't get any new updates (when I tried to report another issue).

@ramiws - Can you confirm - was the fix on the esp32 or the esp8266 and do you have a link to the fix or an IDF or SDK version number which fixes the issue? To be generally useful we need this adopted into the micropython build dependencies.

glenn20, 2023-03-31:
For the information of other readers of this issue... I chased down the bug mentioned by @ramiws above - and yes Espressif provided a hotfix for the espnow library on the esp8266. However, it is for the esp8266 RTOS SDK, not the non-OS SDK used by micropython. It seems they have not responded to a request for a nonOS SDK bugfix.

Unfortunately, I'm not prepared to take on the effort to port micropython to support the RTOS SDK for the esp8266, and I am unaware of anyone else attempting this. So, it seems we are likely to be stuck with this issue for the forseeable future and the use of the AP_IF workaround is the best available solution.

Thanks @ramiws for helping shed some more light on this issue.

ramiws, 2023-03-31:
Just seen this. I was busy porting my firmware to rtos. Yes, they keep updating rtos sdk just to be idf style. Even though no tools or platforms around it. Leaving the nonos sdk that is actually in use. I did binary diff to try patch the nonos espnow lib, but the changes are way more than just the liter fix.

closed 2023-04-02 by glenn20

AmirHmZz, 2023-04-06:
It seems they patched hotfix https://github.com/espressif/esp-now/issues/57 into master branch of nonOS.

glenn20, 2023-04-07:
That's interesting and promising.

Unfortunately micropython uses an old version of the nonos sdk. However it might be feasible that someone can/will do the work to update micropython to the newest NONOS sdk.

AmirHmZz, 2023-09-01:
@glenn20 Have you tested sending data to broadcast address to see if problem exists or not?