glenn20/micropython #11: esp8266 unable to send to broadcast address

larsks, 2022-12-07:
According to [the
documentation](https://micropython-glenn20.readthedocs.io/en/latest/library/espnow.html#broadcast-and-multicast),
esp8266 should be able to send to (but not receive on) the broadcast
address:

> All ESP-Now devices (including ESP8266 devices) can also send messages to the broadcast MAC address or any multicast MAC address.
>
> To send() a broadcast message, the broadcast (or multicast) MAC address must first be registered using add_peer(). send() will always return True for broadcasts, regardless of whether any devices receive the message. It is not permitted to encrypt messages sent to the broadcast address or any multicast address.

However, in practice this seems to fail with an `OSError`:

```
MicroPython v1.19.1-espnow-6-g44f65965b on 2022-07-09; ESP module with ESP8266
Type "help()" for more information.
>>> import espnow
>>> e = espnow.ESPNow()
>>> e.active(True)
>>> bcast = b'\xff' * 6
>>> e.add_peer(bcast)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: -3
```

This is running Micropython installed from <https://github.com/glenn20/micropython-espnow-images/blob/6b9e8d5dd08ee445a9fb3d145e64fa2473239c9a/20220709_espnow-g20-v1.19.1-espnow-6-g44f65965b/firmware-esp8266-GENERIC.bin>.

AmirHmZz, 2022-12-10:
@glenn20 @larsks I've faced the exact same behavior on esp8266 while adding broadcast address as peer. It seems there's no need to add broadcast address as peer in esp8266 but I don't know why and I'm just telling this based on my own experience. So the solution is to forget about adding it as a peer but I think it should be investigated more deeply.

glenn20, 2022-12-13:
Thanks @larsks for the bug report - and to @AmirHmZz for the answer. Yes - on the esp8266 it is not actually necessary to add_peer() devices (though the docs say it is). So, the work-around is to not call add_peer() for the MAC address on the esp8266. 

I will look into it, but the version of the SDK used by micropython is no longer supported by Espressif and has some well known limitations. This may be one fo the less well-known ones.

closed 2023-04-02 by glenn20