glenn20/micropython #17: Library not found in local micropython-lib: aioespnow

cyberedgelord, 2023-04-29:
When trying to compile espnow-g20-v1.20.0 branch, following error pops out:

```
freeze error executing "boards/GENERIC_1M/manifest.py": Error in manifest file: 
/micropython/ports/esp8266/boards/GENERIC_1M/manifest.py: Error in manifest file: 
/micropython/ports/esp8266/boards/manifest.py: Library not found in local micropython-lib: aioespnow
```

I checked the _lib/micropython-lib/micropython_ folder and aioespnow is not there. So, I created the folder "aioespnow" and added "aioespnow.py" from v1.19.1 and everything compiles successfully.

[aioespnow.zip](https://github.com/glenn20/micropython/files/11359997/aioespnow.zip)

glenn20, 2023-05-01:
Thanks for reminding me about this. Yes, we need to wait for `aioespnow.py` to appear in micropythonlib.

I **should** have left aioespnow.py out of the manifest in the branch, but accidentally committed that when I was building the v1.20.0 pre-compiled firmware images (as I wanted to freeze aioespnow into those images).

glenn20, 2023-05-01:
OK - I have reverted the commit to add `aioespnow.py` to `manifest.py` and pushed a new commit to put `aioespnow.py` back into the `esp32/modules` and `esp8266/modules` directories in the `espnow-g20-v1.20.0` branch. I will close the issue, but feel free to re-open if there are any problems.

closed 2023-05-01 by glenn20